FABtotum/fabui-colibri #123: fabui internal error after update

christopherwit, 2017-03-29:
As soon as I update from 0.99.20170327.1753 to 0.99.20170329.1746 the complete webinterface is broken and end up with a 500 internal server error.

When I try to recover with a latest fresh image install I get a bunch of cp, md, read and input/output errors during first boot after initial install wizard.
When I try to access FABUI I get the following error message:
http://169.254.1.2/fabui/install
SQLSTATE[HY000] [10] disk I/O error
Filename: core/CodeIgniter.php
Line number: 500

krios-fabteam, 2017-03-30:
fixed with the new update and with the [new sd card archive](http://update.fabtotum.com/colibri/armhf/images/sdcard_latest.zip)
Once you confirm it is fixed, please close this issue.

christopherwit, 2017-03-30:
Yep everything back up and running again

closed 2017-03-30 by christopherwit