FABtotum/fabui-colibri #126: After autobed leveling no heating of bed

casuda, 2017-04-02:
When I start a print with auto bed leveling and the probing of the bed finished, the print stucks.
The bed does not heated and I have to restart the print again without auto leveling.

krios-fabteam, 2017-04-03:
could you update your unit with the today's update and try it again? 
(ps: remember to reboot the unit after the update task is finished)

casuda, 2017-04-04:
I can't upgrade the fablin it doesn't work...

casuda, 2017-04-04:
on both FABs V1 and V2.... the update procedure starts and say finish, but after an reboot or shutdown and start the Fablin 1.0.0097.2-rc3 was shown.... to upgrade

krios-fabteam, 2017-04-05:
Wich firmware in installed on your units?
Could you paste here "M503" output from "Maintenance>System Info" ?

casuda, 2017-04-05:
On Core V2
Druckereinstellungen (Ausgabe M503)
Steps per unit
M92 X72.58 Y72.58 Z2133.33 E1524.00
Maximum feedrates (mm/s)
M203 X550.00 Y550.00 Z15.00 E12.00
Maximum Acceleration (mm/s2)
M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration
M204 S4000.00 T100.00
Advanced variables
M205 S0.00 T0.00 B20000 X20.00 Z0.40 E1.00
Home offset (mm)
M206 X0.00 Y0.00 Z0.00
PID
M301 P20.00 I3.50 D30.00
Servo Endstop
R: 44 E: 147
Z Probe Length
-38.00
Installed head ID
2
Batch number
5
Baudrate
250000
FABlin version
1.0.0097.2-rc2

krios-fabteam, 2017-04-05:
Let's split the 2 issues.
Could you also paste here the version of "FABUI" bundle?
If your unit was able to update fabui bundle with the installed firmware (1.0.0.0097.2-rc) you should start  a print with auto bed leveling. Can you confirm it? If yes you can close this issue and you can open a new one for the firmware update problem

closed 2017-04-26 by krios-fabteam