FABtotum/fabui-colibri #127: fabLIN 1.0.0097.2-rc3 won't install

lahines100, 2017-04-07:
FABlin 1.0.0097.2-rc3 won't install. After many attempts and restarts FABlin 1.0.0097.2-rc2 remains.

RobertsVancovics, 2017-04-07:
Having same issue.

krios-fabteam, 2017-04-10:
And what about the other versions? 
Can you try to install other different version (for example 1.0.0097) by going in "Maintenance > Firmware" (your unit must be connected to interntet)

RobertsVancovics, 2017-04-10:
I am sorry, I made a mistake. I was having problem installing lates Colibri that day, not updating FABlin.

krios-fabteam, 2017-04-10:
Don't worry ;)
@lahines100  what about you?

RobertsVancovics, 2017-04-10:
But still, was anyone else experiencing problems while installing frydays Colibri? For me everything was fine until I clicked on the button "next". I think it was fourth step (on it was the option to assign a name for my machine and enter its code). Nothing was happening. I waited for 10 min. Reflashed the sd card and tried again with no sucsess. Any Ideas what it could be? At the moment I had slightly modified firmware if that changes anything.

krios-fabteam, 2017-04-10:
Today after 18.00 pm (GMT+1) we'll release a new sd card archive, please try with that.
We also have noticed some issues
Thanks guys

RobertsVancovics, 2017-04-10:
Will try it.

lahines100, 2017-04-10:
I installed the latest core 0.6.7.
After two attempts to install FABlin 1.0.0097 without success I rebooted 
the printer and using Maintenance/ firmware successfully downloaded and 
installed FABlin 1.0.0097.2-rc3. Checking updates shows system up to 
date.
I then attempted to install the V2 print head. The install process 
continued for several minutes with out ever completing. I used emergency 
stop button and saw on the info bar that the V2 print head was 
installed.
I moved to First Setup, was able to skip head install, and moved to bed 
setup. at the second bed setup step I needed to stop the First Setup 
Tour in order to move on the bed leveling process. On completion of bed 
leveling, I could not continue on First Setup tour and had to manually 
move to nozzle calibration. All else seems to work at this time.

<mention@noreply.github.com>
Sent: 4/10/2017 4:38:49 AM
install (#127)

christopherwit, 2017-04-10:
With today update I cannot reproduce this issue. Fixed?

lahines100, 2017-04-10:
Do I need to start new threads for the additional issues mentioned in my 
reply?

Sent: 4/10/2017 3:44:16 PM
install (#127)

krios-fabteam, 2017-04-11:
@lahines100 yes, please.

christopherwit, 2017-04-13:
hey we are up to a new firmware, but I cannot flash fablin 1.0.0098 successfully. I', stuck at 1.0.0097.2-rc3

daniel-fabteam, 2017-04-13:
Wrong firmware was uploaded. Try to update it again. It should show the right version.

christopherwit, 2017-04-13:
Great, thanks for the quick fix, everything is back and good to go.

closed 2017-04-19 by daniel-fabteam