FABtotum/fabui-colibri #129: Connecting to the FabUI does work

daniel-fabteam, 2017-04-10:
(Connecting to the FabUI does work, so the static addresses of the FABtotum haven't been changed by the update. )

Settings -> Network -> DNS shows:
Head: (empty)
Current: 8.8.8.8
Tail: 8.8.8.8
8.8.4.4

daniel-fabteam, 2017-04-10:
@homahe Could you check what the default gateway is when there are network issues?
Connect over ssh and type `ip route list` and paste the result here.

daniel-fabteam, 2017-04-10:
Try to make a new sdcard with todays update.

homahe, 2017-04-10:
Hello, Daniel,

ip route list shows:
192.168.17.0/24 dev eth0  src 192.168.17.171

As I can ping the gateway via the FABtotum (and DNS does work correctly using the PC controlling the FABtotum) I'd guess that /etc/resolv.conf is the culprit, because the name-resoloution doesn't work, when I try to ping an address.
Besides, Settings -> Network -> Network-Settings read:
Address mode  sttic
Address  192.168.17.171
Netmask  255.255.255.255.0
Gateway  192.168.17.1

(Hint:  Accessing the FABtotum via SSH /root may be considered as an invitation ;-) )

jm2¢

daniel-fabteam, 2017-04-11:
How is your wlan0 configured? (`Disabled`, `Static`, `DHCP`)
Are you using `static` or `DHCP` for eth0?

You can try to execute `ip route add default via 192.168.17.1 dev eth0` and see if internet starts working.

PS.: ssh access will get a password in the full release.

homahe, 2017-04-11:
I'm using static addresses for eth0 and have no WLAN-adapter installed.

Having installed FabUI 0.99.20170410.1750 / FabLin 1.0.0097.2-rc3 from
scratch,

using the method to change the network-settings in /earlyboot/earlyboot.conf

...

# Network settings
NETWORK_IF="eth0"
NETWORK_IPV4="169.254.1.2/16"
#NETWORK_MASK="255.255.0.0"
NETWORK_GW="169.254.1.1"

to

# Network settings
NETWORK_IF="eth0"
NETWORK_IPV4="192.168.17.171/25"
#NETWORK_MASK="255.255.255.0"
NETWORK_GW="192.168.17.1"

before FabUI installation was working now without falling back to the
default values and DNS was retained. Very good!

If you login via ssh, a change to a preferred DNS address to, say
kind'o  OpenDNS e.g.

  * 204.152.184.76 (f.6to4-servers.net, ISC, USA)
  * 2001:4f8:0:2::14 (f.6to4-servers.net, IPv6, ISC)
  * 194.150.168.168 (dns.as250.net; anycast DNS!)
  * 213.73.91.35 (dnscache.berlin.ccc.de)
  * 80.237.196.2
  * 194.95.202.198

in /etc/resolv.conf with editor /nano/ does work too.

Using head or tail will not get rid of google, so my experience, but add
the addresses to nameserver 8.8.8.8, as far as I know, only the first
three addresses in the .conf-list are used.

closed 2017-06-28 by krios-fabteam