FABtotum/fabui-colibri #130: Settings saving doesnt work

RobertsVancovics, 2017-04-10:
When I am trying to save z max position through Settings>Hardware>Homing>Z max home position, my saved settings are replaced by default height even after restarting right after saving- I have ticked the "use custom settings". The default height is 241,5mm , but I want it to set as 234,6mm(I have a custom head). I would like to calibrate my nozzle height, but I am a bit scared, that by calibrating the neozzle height with the assisted calibration my hotend will slam straight into the heating bed. Do I have to worry about that?
Also, thank you so much for creating an easly accesable and editable G-code folder! (I have only messed around with T0 and T1 comands)

krios-fabteam, 2017-04-11:
You don't have to worry
Z max is handled as fixed value in fabui-colibri (we will hide it from settings to avoid any further confusion)
with the assisted nozzle height calibration you should be able to measure the offset of the nozzle of your custom head, try it being careful and if you see that the nozzle is about to crash the heated bed click immediately the emergency button and please report it here

RobertsVancovics, 2017-04-11:
I callibrated z heigth sucsesfully. there was a good 20 mm gap between the nozzle and hotend during calibration, so nothing to worry about.

closed 2017-04-11 by krios-fabteam