FABtotum/fabui-colibri #134: extruder stops after first hour

mjb984, 2017-04-15:
unable to print any item that take more then an hour. the printer continues to function but stops extruding shortly after an hour. the filament seems to be grinded and I have attempted various fixes recommended by others including manually forcing the filament if it gets stuck but that only works for a few minutes, changing the retraction settings on the slicer, different filaments PLA and ABS (have done so but only with fabtotum filament) even disabling the slicer completely. 

please help as I have been unable to print anything bigger than a sugar cube.

regards,

RobertsVancovics, 2017-04-15:
It could be that you are printing too fast or your temperature is too low. Have you tried tweaking those?

krios-fabteam, 2017-04-18:
Hi @mjb984, 
this is not an issue related to the software, for any kind of issue with the hardware please open a [ticket](http://www.fabtotum.com/support) and our support team will help you

mjb984, 2017-04-22:
i have tried both and other suggestions i have found on your forum with no
success. in addition to this i have more things that dont work such as
the scanner. none of the options work with the exception of sweep. and the
format generated is only viable in mesh lab but not usable.
the ABS filament i purchased from you has horrible adhesion and when is
does stick the always gets stuck in the nozzle and extruder.

the filament gets clogged all the time. regardless of filament. i have to
clean the nozzle and have even noticed filament stuck in the hot end. i
have done all steps and followed all directions offer on your site again
with no success. even went as far as vacuum sealing the rolls in

the nozzle tip seems like the hole is offset from the moment i received it.
If so this is a huge inconvenience for various reasons. your nozzle cost me
a lot of time and money . your the only company that does not support the
market standard sizes available to me locally. and if i where to return the
head or even just the nozzle it would cost me as much as a new head after
considering what shipping and the taxes i have had to pay twice already.

the 3d head is a nightmare. i have been unable to complete any print job.
not even the ones preinstalled in the interface. the most hurtful part is
that it seems like you have encountered and possibly corrected some of the
issue after i purchased and received my device in the new PRO head. which
means i have to buy a new accessory to make my device work as it should
have to begin with.

the support you offer is no different then the information offered on the
website which has not helped at all. there is no support on the use of the
device either. i have had to figure most of it out on my own and with the
help of your disgruntled followers on the forum that seem to be
experiencing the same issues.

your devices is the only one in this price range that is not supported by
any of the usual fabrication software directly. and the configuration files
(work around) you offer do have any instruction on configuring them.

not all has been a disaster. the leaser head itself has proven usful but
the configuration and file creation are inconvinient. the app while light
wait requires internnet which i have issues in my area with at times so new
engravings are difficult. the predefinded profiles are vaige and not
friendly for people like myself new to laser engraving. the power toggle
does not work, and its very tedious to have to make an image, pass it to
the web site, then download it, then load it up to the fabtotum. this
should be something much more stream line in my opinion. in addition the
scanner( if it works) should have the ability to help properly align the
beam and even an interface to assist in centering the laser engraving
design as do many other devices at lower price points.

Not that your intrested but this has left such a back taste that i have
lost all intrest in the machine and 3D printing in general . i just refuse
to get taken advantage of. i would like to ask for your suggestion to
resolve this issue. as far as configuration changes and updates i have
exhausted them as have many of your customers. so please dont insult me by
offering any more. i will continue to keep an openmind on this matter but
only if i am treated respecfuly. not like a guinipig. becuase it seems like
these things should have been worked out before offering the device or
offered at a much more competitive price point with the disclaimer that its
a beta testing stage.

closed 2017-05-30 by daniel-fabteam