FABtotum/fabui-colibri #137: some problem

PaoloZa, 2017-04-17:
Hi 
I test colibri for the first time, some problems that I find:

The raspi Cam is not detected.
The nozle appair injecting plastic excessive, creating lumps and deposits, it does not happen with the same files in the original software.
The button "canceling printing" sometime doesn't respond.
I have printer Core edition .
Paolo

daniel-fabteam, 2017-04-18:
Hi Paolo,

When you aborted the print. Did you do it during the nozzle heating phase or when the print already start (head movement)

Could you install "System logs" plugin and attache the logs here?
Thanks

christopherwit, 2017-04-18:
Hi Paolo,

Which web browser do you use? 
In firefox my raspi cam is detected successful.

PaoloZa, 2017-04-19:
Hi

I aborting during the printing not in the nozle pre heating.
I had change the page of software menu during  the print progress, return to the printing status page and I aborted the print, the commnad doesn't respond.

I use Firefox 52.02 ,I go to hardware setting > cam setup, the page show me The Raspberry camera doesn't detect.

PaoloZa, 2017-04-19:
After my test I change the SD-flash and return to the FabUI v. 0.987 original software.
The print doesn't work, I had to repeat all calibration of bad and nozle, the sotwares change a configuration of firmware ?

igough, 2017-04-23:
I am also experiencing the problem with excessive material being layed down. I was attempting to build the rpi2-bottom_netfabb.stl model from http://www.thingiverse.com/thing:922740. This model printed reasonably well prior to changing to Colibri. During the first circuit of the brim it seemed like the feedrate just slowed down by 10x or so during the corners, laying down large blobs of PLA. The straight lines between the corners were layed down fine. The machine was also occasionally making alarm sounds. I thought the head might be hitting the glass, dragging the head and causing the slowdown, so I jogged the bed down 0.1mm, but this made no difference other than preventing the PLA from adhering to the bed during the straight sections. After I aborted, the UI complained that no head was installed (it lies, I went through the install procedure for the hybrid head, and double checking showed the Hybrid head was still showing as being installed). It also complained that there was something wrong with the bed. I did not do the first time setup, but rather manually ran all of the calibrations except for bed levelling, since the bed is levelled to where it works best (using the old software), and bed levelling makes it worse.

[image]

daniel-fabteam, 2017-04-24:
@PaoloZa  @igough   Could you post which unit versions you have?

igough, 2017-04-24:
Os - Linux fabtotum 4.4.38-v6 #1 Fri Feb 24 12:02:46 CET 2017 armv6l GNU/Linux 
Fabui - 0.99.20170420.1718
Firmware - 1.0.0098	- Build date: Apr 12 2017 14:39:36	- Author: FABteam
Hardware - 1
Installed head - Hybrid Head

igough, 2017-04-24:
Core - v0.6.8 Build date: 2017-04-14
Kernel - v0.6.1 Build date: 2017-02-24
Web - v0.6.0 Build date: 2017-02-23
Libs - v0.6.0 Build date: 2017-02-23
Python - v0.6.0 Build date: 2017-02-23
Python-extra - v0.6.0 Build date: 2017-02-23
Network - v0.6.3 Build date: 2017-03-17
Scientific - v0.6.0 Build date: 2017-02-23
Tools - v0.6.0 Build date: 2017-02-23
Rpi - v0.6.0 Build date: 2017-02-23
Samples - 0.6.1 Build date: 2017-04-06
Fabui - 0.99.20170420.1718 Build date: 2017-04-20

daniel-fabteam, 2017-04-24:
Could you attach the output of `M503` and the gcode file you were using? So we can try and replicate the issue.

igough, 2017-04-24:
Steps per unit - M92 X72.58 Y72.58 Z2133.33 E3048.16
Maximum feedrates (mm/s) - M203 X550.00 Y550.00 Z15.00 E12.00
Maximum Acceleration (mm/s2) - M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration - M204 S4000.00 T100.00
Advanced variables - M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm) - M206 X0.00 Y0.00 Z0.00
PID - M301 P15.00 I5.00 D30.00
Servo Endstop - R: 51 E: 149
Z Probe Length - -38.00
Installed head ID - 1
Batch number - 1
Baudrate - 250000
FABlin version - 1.0.0098
[gcode.zip](https://github.com/FABtotum/fabui-colibri/files/952794/gcode.zip)

daniel-fabteam, 2017-05-02:
@igough Remove `G28` from your start gcode. All the axis are already homed and nozzle offset is adjusted before the gcode file is executed.

igough, 2017-05-02:
I assume you are suggesting that rehoming is not as efficient, not that it is causing the slowdown in the corners that causes the excess material to be emitted.

daniel-fabteam, 2017-05-03:
`G28` messes up the adjustments made before the print starts. That would explain why the nozzle is hitting the glass. I've used you gcode and once I removed `G28` it printed well.

[image]

After abort/emergency is normal that the head/bed are not detected as the back-end services are restarting. After few seconds the head/bed check should show the correct values once the services are back.

igough, 2017-05-03:
Just tried removing the G28 and it prints fine now.

I have removed that G28 from the start Gcode in the Cura profile for the Fabtotum. I checked the start Gcode at [https://github.com/FABtotum/FAB_Configs/blob/master/Cura2_configs_FABtotum/Cura%202.5/resources/definitions/fabtotum.def.json](url) and it is totally different from what I have installed, but I am running Cura 2.4 and not 2.5. Are there resources somewhere which recommend the startup Gcode for Cura 2.4?

daniel-fabteam, 2017-05-03:
Questions related to usage are best addressed on the forum. There you should get a faster response from the support team and even other users. 

As the issue turned out not to be related software but configuration I'll close it.

closed 2017-05-03 by daniel-fabteam