FABtotum/fabui-colibri #139: Raspicam not detected

christopherwit, 2017-04-18:
Currently the settings-raspicam page doesn't detect raspicam in non of the following browsers:
firefox, edge, IE11 and google chrome.

krios-fabteam, 2017-04-19:
Hi @christopherwit , 
could you paste here M503 output (you can get it in Maintenance > System Info)

christopherwit, 2017-04-19:
Hi guys in latest release 0.99.20170419.1714 I do get camera visuals when I use detect button in the no camera detect message.

daniel-fabteam, 2017-04-20:
@christopherwit After using the `Detect` button, does the camera stay detected later?

christopherwit, 2017-04-20:
I'm not sure after a restart, but within the same browser and session the
camera stays detected.
Maybe I can test a bit further this evening. I'll keep you updated.

christopherwit, 2017-04-20:
Hi daniel, yes it still works when detect is selected even after a full power down and restart.
Which made it only an minor issue that it doesn't work right out of the box :)

daniel-fabteam, 2017-04-24:
The detection is done during first boot so the issue should not be there on a fresh install and after full release.

closed 2017-04-24 by daniel-fabteam