FABtotum/fabui-colibri #141: Nozzle offset value missing from M503 output

christopherwit, 2017-04-19:
I do miss the result of nozzle height fine calibration - nozzle offset in the M503 output.
Is it actually a value that can be set with a Mcode?

I would really like to store this value in the custom gcode section for the different heads I have similar to the PID that can be stored.

daniel-fabteam, 2017-04-19:
The nozzle offset is stored for every head individually so there is not need to do it manually by gcode.

If you do `Export` of the head settings you will find the value in the resulting json file.

christopherwit, 2017-04-19:
Thanks for letting me know, I wasn't aware of this setup. 
In that case I'm still missing the value of the nozzle offset in the M503 output.
And the head export states a 0 nozzle offset no matter how I calibrate, which makes me understand why I have to recalibrate everytime I swap heads.

christopherwit, 2017-04-19:
I notice fabui stores the value somewhere as it is retrieved on head reinstallation. But head settings export shows nozzle offset 0 still.

daniel-fabteam, 2017-04-20:
The missing `nozzle_offset` attribute in exported json file will be fixed in the next update.

closed 2017-04-25 by daniel-fabteam