FABtotum/fabui-colibri #150: Feeder calibration issue

homahe, 2017-04-21:
After having updated to
Os     Linux fabtotum 4.4.38-v6 #1 Fri Feb 24 12:02:46 CET 2017 armv6l GNU/Linux
Fabui     0.99.20170420.1718
Firmware     1.0.0098 - Build date: Apr 12 2017 14:39:36
on my  Hardware   3    using  Printing Head V2
I noticed, that there must by an unexpected change in the feeding amount as the print started to smear.
So I re-calibrated the feeder. 
The stored value fed 172 mm  instead of 100 mm. After having entered the actual length measured I had the feeder parameter recalculated. As the new value has more than sensible decimals I altered 
xxx,1623743211234   to  xxx,2    as new value and then clicked calibrate again.

This time the feeder RETRACTED the filament about 161 mm instead of feeding it.

I could repeat that behavior, so I tend to name it a bug

daniel-fabteam, 2017-04-24:
There are a few `v3` units that have a feeder that was introduced in `v4` batch so you can try this fix.
It will change your unit's batch ID from `v3` to `v4`.

Execute these codes in jog console

    M763 S4
    M500

After this do an `emergency stop` or `reboot`.

Feeder steps for `v4` are `1524` so it should fix your issues #153, #151 too.

If this helps, close the issues.

homahe, 2017-04-24:
All right, Daniel, 

I've done, what you recommended, but that altered only the default value. When I abort a print, the value that was the result of the of the calibration is lost. 

Besides, if you enter the correct value in the Feeder Profile  and use a decimal point instead of a comma, it will be stored without comment, but when you check it, it will be zero.

daniel-fabteam, 2017-04-25:
The lost calibration will be fixed in the next update.

However the dot/comma issue resulting in a zero I cannot replicate. Have switched my system to German to have comma as decimal separator and tried comma AND dot in every feeder view but the values got stored correctly.

homahe, 2017-04-25:
I'm using the English version, however  when in nozzle distance
calibration I have to use the localized comma as defined in the Linux
environment, whereas in the feeder calibration the behaviour (in the
previous version, I updated this morning) was as described.

This might be due to that in one instance the format is determined by
the settings in the OS the browser is running on, in the other case
controlled by the app.

LeiEng, 2017-05-09:
I have the same problem, and as far as I remember I have the v1 hardware (but "System Info" in Colibri OS reports "**Hardware** 5").

If I do a M503 using the original FABtotum software (not Colibri OS), I get:
  Steps per unit: M92 X72.58 Y72.58 Z2133.33 **_E3048.16_**

If I do the same thing when using Colibri OS (Fabui 0.99.20170503.1639, Firmware 1.0.0098 - Build date: Apr 12 2017 14:39:36) I get:
  Steps per unit: M92 X72.58 Y72.58 Z2133.33 **_E1524.00_**

So when I use Colibri OS and feed 100mm of filament, my FABtotum only feeds about 50mm of filament. When I feed 100mm of filament using the original software, 100mm of filament is fed.

Any suggestions what I should do to get this right?

krios-fabteam, 2017-05-09:
Could you paste here the complete M503 output both from fabui-colibri os and the older FABUI ?
Are you using "custom" or "default" settings? (menu: _Settings >Hardware_)

In fabui-colibri then go to jog panel and on mdi input send this:
`M763 S1`
`M500`

after that press "_Emergency_" button (top bar) and check again M503 output

LeiEng, 2017-05-09:
Here are the results:

Old FABUI:
M503: FABtotum TOTUMDUINO
 Batch Number: 5
FABlin
 Version: V 1.0.0098
 Baudrate: 250000
Steps per unit:
  M92 X72.58 Y72.58 Z2133.33 E3048.16
Maximum feedrates (mm/s):
  M203 X250.00 Y250.00 Z15.00 E23.00
Maximum Acceleration (mm/s2):
  M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration
  M204 S4000.00 T100.00
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
  M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm):
  M206 X0.00 Y0.00 Z0.00
PID settings:
   M301 P15.00 I5.00 D30.00
Servo Endstop settings: R: 36 E: 142
Z Probe Length: -25.13
Installed head ID:
 M793 S1
ok

FABUI-Colibri:
M503
FABtotum TOTUMDUINO
 Batch Number: 5
FABlin
 Version: V 1.0.0098
 Baudrate: 250000
Steps per unit:
  M92 X72.58 Y72.58 Z2133.33 E1524.00
Maximum feedrates (mm/s):
  M203 X550.00 Y550.00 Z15.00 E23.00
Maximum Acceleration (mm/s2):
  M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration
  M204 S4000.00 T100.00
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
  M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm):
  M206 X0.00 Y0.00 Z0.00
PID settings:
   M301 P20.00 I3.50 D30.00
Servo Endstop settings: R: 36 E: 142
Z Probe Length: -25.13
Installed head ID:
 M793 S2
ok

I use "default" settings in "Settings > Hardware")

FABUI-Colibri after doing "M763 S1", "M500" and pressing "Emergency" button

M503
FABtotum TOTUMDUINO
 Batch Number: 1
FABlin
 Version: V 1.0.0098
 Baudrate: 250000
Steps per unit:
  M92 X72.58 Y72.58 Z2133.33 E1524.00
Maximum feedrates (mm/s):
  M203 X550.00 Y550.00 Z15.00 E23.00
Maximum Acceleration (mm/s2):
  M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration
  M204 S4000.00 T100.00
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
  M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm):
  M206 X0.00 Y0.00 Z0.00
PID settings:
   M301 P20.00 I3.50 D30.00
Servo Endstop settings: R: 36 E: 142
Z Probe Length: -25.13
Installed head ID:
 M793 S2
ok

krios-fabteam, 2017-05-09:
Now go to Maintenance > Feeder>Profile and set the correct value for step per unit, then you should  have the right value on M503

LeiEng, 2017-05-09:
Yes, that did it!

And I can switch between the old FABUI and the new Colibri-FABUI and they both report:
Steps per unit: M92 X72.58 Y72.58 Z2133.33 **_E3048.16_**

Thanks for your help!

closed 2017-05-10 by krios-fabteam