FABtotum/fabui-colibri #153: feeder parameter reset and unexpected consequences

homahe, 2017-04-24:
Whenever one aborts a print either by 'aborting' it in the dialogue or by opening the front door when the safety option is active, the feeder step parameter is reset to 3048,16. 

If now you got wise and use 'Maintenance', do a  Nozzle height calibration (just to be  sure), then go to 'Feeder' -> 'Profiles' -> having the 4th axis combo: Enter the correct value here changing from 3048,16 to 1584,2, do 'save' and start a new print via object manager, of course you will not expect what will happen:

When head and bed temperature is reached and print start is signaled, the filament is retracted.

Besides: When a pause or an abort is triggered it would make sense to retract the filament not to leave blobs and whiskers on the print...

daniel-fabteam, 2017-04-24:
Could you repeat the scenario and add the `system logs` ?

daniel-fabteam, 2017-05-03:
Was this issue resolved with the recent updates?

closed 2017-05-30 by daniel-fabteam