FABtotum/fabui-colibri #161: Can't update firmware

Tischl0r, 2017-04-28:
I'm running my FAB (one of the first version) first time since some months and installed on a SD card Colibri.
Installation is OK. If I check updates, there is an firmware update.
Installed version: "null", online version: 1.0.0098

If I press "Install" it says "connecting to update server" and thats it.
"Show me details" only shows me an empty line.

What should I do?

daniel-fabteam, 2017-05-03:
Can you check if your unit has internet? There should be a small planet icon at the top next to the menu.

Also how is your unit's network configured?

Tischl0r, 2017-05-03:
Yes, internet is avaiable. It shows me new feeds on the dashboard and the planet icon is avaiable.
The FAB is via Wifi connected to my router (and internet), DHCP is activated.

krios-fabteam, 2017-05-04:
Could you try to flash the firmware by going to "Maintenance > Firmware" and selecting the latest version (currently is 1.0.0098)

Tischl0r, 2017-05-04:
Nope, not possible...
I select firmware, click on "flash firmware". It appears a window "installing firmware 1.0.0098..." and a "waiting-circle" appears. After some seconds, the window refreshs and i'm back in the Maintenance > Firmware window. Still all fields on the right side "n.a."

daniel-fabteam, 2017-05-05:
Could you install the `System Logs` plugin and attache the logs here?

You can find the `System Logs` plugin under `Plugins > Online`. Click on `Install` and then on `Activate`.
You will get a new menu item. Click on it  and then on download. It will take some time to compact all the logs.

Tischl0r, 2017-05-05:
Sorry, it doesn't works.
I click on "Install", the button changes to "Connecting..." and after some seconds the page refreshes and i'm back in the Plugin Main Menu, where it says "no plugins installed"

daniel-fabteam, 2017-05-05:
You can try to download the plugin from [here] (https://github.com/FABtotum/fab_system_logs/archive/0.10.0.zip) and then go to `Upload`.

Tischl0r, 2017-05-07:
Puh...
I select the zip file, press upload, and after a while it says in a red window at the right corner:
error
uploaded zip is not a plugin

Tischl0r, 2017-05-07:
By the way - I recognized that the yellow light from the fabtotum non stop is fading.

daniel-fabteam, 2017-05-08:
We are working on a diagnostic plugin to help pinpoint problems easier.

This version supports firmware diagnostics.
[fab_diagnostic-master.zip](https://github.com/FABtotum/fabui-colibri/files/984587/fab_diagnostic-master.zip)

Try to install it. After activation there will be and a new menu item `Maintenance > Diagnostics`.

Go to `Firmware` tab and run tests. Only test that should fail is the 115k baud rate test.

Tischl0r, 2017-05-08:
Same problem... Can't install it

daniel-fabteam, 2017-05-09:
Could you try to make a new sdcard again by downloading the latest image?

Tischl0r, 2017-05-09:
Hi!

I downloaded the latest image now and yeah, it works!
The firmware is the latest and I could install plugins.
And no non stop fading yellow light, it stays after booting at white.

Thank you!

closed 2017-05-10 by krios-fabteam