FABtotum/fabui-colibri #163: Grinding filament

AndrewKillian, 2017-04-30:
Recently, the feeder has been grinding up filaments that previously have printed with great results. After a while of printing, the printer will stop extruding and upon removing the filament it is very obvious that the filament has been ground down to the point where it can not be fed further anymore. I've attached a picture to show what I mean.
[image]

homahe, 2017-05-01:
Hello Andrew,
I noticed the same problem recently too. I've had the impression, that it depends first on the model you're going to print. If there are many retractions in a short sequence it happened; I was printing a set of four cylinders of just 8 mm diameter with a 4.2 mm bore. Second I noticed that the temperature which was set to 205° centigrade went down to 195 ° before the heater caught in ( I guess it was down even lower...)
Third: Because of the picture you provided the pressure in your feeder (I think you've got a head type II) is too high. (The pressure is fixed now, whereas it was adjustable before. The tensioning spring is longer and the nut is left out in the feeder, wonder why there is screw at all...) How the filament should look like, when the pressure is correct you can see here:

https://ultimaker.com/en/resources/21477-how-to-fix-under-extrusion

Of course it should be considered, what kind of filament is used and if it is PLA how it has been stored.
Sometimes it is recommended to dry the filament by heating it for two hours to 80° in a ventilated oven.

If the filament shows a relative wide flat surface opposite the tooth-marked side, the pressure is considered to be too high. If there are kind of regular marks, where there the tooth marks are not clearly visible that might be due to residues of filament flakes in the feeder. 

Sorry, but cleaning the feeder and the tube IMHO is somehow recommending to consider to purchase the Head_pro, not currently available ;-).

jm2¢

AndrewKillian, 2017-05-03:
This appears to be a mechanical issue so I'll close it for now

closed 2017-05-03 by daniel-fabteam