FABtotum/fabui-colibri #164: Bed leveling

AndrewKillian, 2017-05-03:
the bed leveling seems to not be working correctly as when I calibrate it claims all points are well done however when printing it is obvious that the bed is not leveled. I calibrate with two pokes and that normally gives me some feedback but after I change it, it still says well done even when the bed does not seem properly leveled.

homahe, 2017-05-03:
Hi Andrew,

there are several possibilities that might cause this, as I've found out.

As I've reported in an issue, when the a procedure was interrupted by opening the front-door and the security flag is set, in a bed levelling procedure thereafter signals 'Well done' even when the probe did not reach the plate. And that is true even when you repeat the task. So there might be the possibility, that this was the case here, but that would be very uncommon. 

More common are mechanical problems:
Even when the probe, the needle, is calibrated to switch out to perfect 90°, it tends to slip on the glass-bed. (Besides, either the 90° are true relative to the frame, but not to the glass-bed when it is not perfectly level to the frame, or the opposite is the case. In the first case there is a resulting force causing the needle to slip sideways in the second case there is a difference in the way the probe-needle will have to travel...)
To prevent the error caused by this, I try to calibrate the probe to be as exact at 90° relative to the frame as possible and use a bit of hair-spray on the glass-bed to minimize slipping where the probe touches the surface.

And there is another mechanical problem. Mathematically a plane is defined by three points, but here we have four fixtures. And the fixtures are realized by magnets. When the bed sits on three points the fourth might be pulled up instead of sitting on the thread of the support. And, as there is horizontal play in the two supporting arms, a magnet might even not sit in the bore of the bed. 
Therefore it is recommendable to check if the magnets fit correctly and push down the bed to check for tolerances. 
Now, taking one point as fixed (here it is N° 2)  and the bed not elastic and plane, the load of the probe negligible, being asked to correct at one point, that can be interpreted as a measure of inaccuracy of the process. So a correction of about that value might be necessary at one side (two points, namely 1 + 4 or 3 + 4 in this case) to level the bed perfectly.

jm2¢

daniel-fabteam, 2017-05-03:
Will be fixed in the next update.

homahe, 2017-05-03:
having read your post again, I see that I haven't taken into account, that you deliberately have altered the bed adjustment. So another reason comes to mind:
As stated, when the probe never reached the bed, 'Well done' is signalled. That might be true too, when the head is not seated correctly, has a contact problem, or the probe is defect, as long as the software does not check for a 0/1 transit before starting the measurement.

homahe, 2017-05-03:
The bed levelling in the laser-head / bed reversed version did work for me.

closed 2017-05-10 by krios-fabteam

Mnemo80, 2017-06-02:
I think i found the solution.   After the update re-enable the probe testing before the print.  I fixed it in this way. Then i turned it off again and all works well.