FABtotum/fabui-colibri #166: no printhead in make print

nicolasvandenplas, 2017-05-04:
after installing colibri software and do first setup.
where all is going fine, (bed calibration, head installing , spool load , calibrating bed, )
in make I go to print and it do not see any printer head.
in software it say that the printer head is installed.

krios-fabteam, 2017-05-04:
Hi, 
could you paste here a screenshot of what you see when you go in "_Make > Print_" ?
and could you also paste here "M503 Output" (you can find it by going to "_Maintenance > System Info_")

nicolasvandenplas, 2017-05-04:
[image]
[image]
[image]

nicolasvandenplas, 2017-05-04:
hoop this helps.
greetings Nicolas

krios-fabteam, 2017-05-04:
Your unit is running with an old firmware version, please update it (go to "_Updates_" page or "_Maintenace > Firmware_")
Could you also go to "_Jog_" panel and send the command "M745" and paste here the response?

nicolasvandenplas, 2017-05-04:
[image]

nicolasvandenplas, 2017-05-04:
this do not do anything.
and the firmware is also not downloading .
after download it still say that the old firmware is on

nicolasvandenplas, 2017-05-04:
[image]
[image]

nicolasvandenplas, 2017-05-04:
can I find a tutorial on how to do it.
I think I do something bad
and a link to the download page to find it like that?
greetings Nicolas

daniel-fabteam, 2017-05-05:
Could you?
1) install the `System Logs` plugin 
2) try to install latest fw from `Maintenance > Firmware`
3) attache the logs here

You can find the System Logs plugin under Plugins > Online. Click on Install and then on Activate.
You will get a new menu item. Click on it and then on download. It will take some time to compact all the logs.

nicolasvandenplas, 2017-05-05:
[image]

nicolasvandenplas, 2017-05-05:
FW not able to download
system login installed

nicolasvandenplas, 2017-05-05:
[image]

nicolasvandenplas, 2017-05-05:
[archive.tar.gz](https://github.com/FABtotum/fabui-colibri/files/979279/archive.tar.gz)

nicolasvandenplas, 2017-05-05:
can you send me a link  with the zip file that I can install ?

FABui 0.987 software
and to firmware 1.0.0098

it is a week that I can't work with my printer, maybe I put back fabui software

nicolasvandenplas, 2017-05-05:
so that I can put it on my pc and try to upload it from there.

thanks Nicolas

krios-fabteam, 2017-05-05:
Here  you can find the links:
[http://www.fabtotum.com/restore-corrupted-sd-card/](http://www.fabtotum.com/restore-corrupted-sd-card/)

nicolasvandenplas, 2017-05-09:
hello
I did install the FABui soft and the printer is working ok.
I still have the old firmware running , is there a way to update it ?

I also have still the SD card whit the Colibri software on it, hoop that there is a way to make it work.

I see that in the colibri software the printer head is named (printer head light )
and in the FAbui software it is named (printer head V2)

hoop this helps to in proof the software

greetings Nicolas

closed 2018-01-11 by krios-fabteam