FABtotum/fabui-colibri #168: Feeder profiles does not seem to work

LeiEng, 2017-05-29:
If I add a feeder profile in "Mainenance -> Feeder -> Profiles", it does not show up in the dropdown menu below "Please select which feeder you want to configure". However, an extra empty line shows up in the dropdown menu.

I was able to select the empty line and click "Configure", so now my FABtotum tells me:

  Currently your **FABtotum Personal Fabricator** is configured to use

(So it tells me I am using nothing as feeder)
Now it is impossible for me to get back to the original "Built-in feeder". When I select it in the dropdown menu and click "Configure" nothing happens.

I am using ColibriOS Core v1.0.0 built on 2017-05-24.

daniel-fabteam, 2017-05-29:
The fix is available in the latest FABUI update.

LeiEng, 2017-05-29:
Thanks for the quick update!

I have updated to the latest version, but I still cannot get the "Maintenance -> Feeder -> profiles" to work.
I currently have three profiles in the dropdown menu, the first one is blank, the second one is "Built-in feeder" and the third one is blank.
It still tells me I am using nothing as feeder, and if I select "Built-in feeder" and click "Configure", nothing happens.

daniel-fabteam, 2017-05-30:
Try to remove the bad formatted config files. 
Connect over ssh to your unit and remove custom configs. You can find the feeder profiles in `/var/lib/fabui/feeders`.
Only leave `built_in_feeder.json` as that is the config for the internal feeder.

If this does not help. Use the `System logs` plugin to download the logs and attach them here.

LeiEng, 2017-05-30:
I removed the two "faulty" files from `/var/lib/fabui/feeders` and now the "Feeder profiles" page works as it should!
Thanks for the quick fix!

closed 2017-06-06 by krios-fabteam