FABtotum/fabui-colibri #170: fablin 1.1.0 does not get installed by update

christopherwit, 2017-05-29:
Hi Guys,

Finally I had some time left to get on with my fabtotum and noticed all your updates. Congratulations on the release of the first major version of Colibri. 

Unfortunately I'm stuck at fablin 1.0.0098 as firmware version 1.1.0 does not get installed by the automatic updater.

christopherwit, 2017-05-29:
Sorry mo clue what went wrong, but after 8 attempts the system somehow made it till the end. Currently running fablin 1.1.0 with great success.

closed 2017-05-29 by christopherwit

daniel-fabteam, 2017-05-30:
Hi Christoper,

The firmware can always be installed from `Maintenance > Firmware` in case something goes wrong with update.

Is just a tip. Always report an issue when something is not working as it should.