FABtotum/fabui-colibri #172: Bed leveling problems

AndrewKillian, 2017-05-30:
when calibrating a second time the probe will come out and the head will move to each of the four positions but the bed will never actually move up to probe and the calibration never finishes

daniel-fabteam, 2017-05-30:
Have you updated the firmware to 1.1.0?

AndrewKillian, 2017-05-30:
yes

LeiEng, 2017-05-30:
I can confirm that I have the exact same problem. And I have no updates pending (the firmware is 1.1.0).

krios-fabteam, 2017-05-31:
The fix will be available with the new updates,  there will be also a firmware update

AndrewKillian, 2017-06-01:
I have updated the software and now the procedure does not begin at all it just beeps and says well done for all.

Mnemo80, 2017-06-02:
Turn on the probe test before the print in the setting section.  Do the bed leveling and then turn it off the option again if you want.

closed 2017-06-08 by krios-fabteam