FABtotum/fabui-colibri #173: Print bed crashed into the print head at start of print!

LeiEng, 2017-05-30:
Today I finally decided to try to print something with the new Colibri OS. Previously I have just started it and looked around in the menus to see how it looks.

Unfortunately the first print with the Colibri OS went REALLY bad 😞  

It looked OK at first, but I noticed that the printer never extended the probe to do a "Simple homing". I thought that it was a bit odd that I got directly into the page where the printer waits for the temperature to reach the correct values.
When the temperatures finally reached the correct values the printer started to raise the print bed, and unfortunately it kept raising it until the print head crashed into the bed (in the front left corner). This of course dislocated the bed, and unfortunately it also broke my print head 😞  

After removing the print head (which was not easy) and shutting down the printer, I started it up again. After a little fiddling around I could lower the bed again.

After shutting the FABtotum down, reattaching the print bed and attaching another print head, the printer at least started up again. But now I don't dare to try to print anything with Colibri OS again. I don't want to damage another print head, the bed or even worse, the carriage.

How do I proceed with this? I definitely don't dare to try to print again, I don't want to damage my printer more than I already have. But I really want to find out what the problem is and hope that you can find a fix for it.

LeiEng, 2017-05-30:
To make things worse (at least for me), it does not seem to be possible to go back and use the original FABtotum software after the update to firmware 1.1.0.

If I bootup the original FABUI software from another SD card and go into the JOG menu, the current temperature of the bed and extruder is not displayed. Where those values should be it just says "**NaN°C**". And in the heading where the temperature controls are, it says "**N: /0°C - B: /0°C**".

I guess that this means that the printer will never realize that it reaches the correct temperature, is that correct?

PaoloP74, 2017-05-31:
I think firmware 1.1.0 is incompatible with old FabUI, you should go back to version 1.0.0098.1.

krios-fabteam, 2017-05-31:
@PaoloP74 is right, if you want to use the older FABUI you should downgrade firmware to version 1.0.0098.1

LeiEng, 2017-05-31:
Are you sure about the 1.0.0098.**_1_** version? The highest one I can find at http://<my_fab_ip>/recovery/flash.php is 1.0.0098
The same goes for the one in the new Colibri OS, it is 1.0.0098, not 1.0.0098.**_1_**

Another question; when you start a print in the new Colibri OS and select "Simple homing", does i use the probe for this or not? And when does this homing takes place if the probe is used, before or after the printer reaches the correct tempreature?

PaoloP74, 2017-05-31:
Latest FabLin release for old FabUI is here: https://github.com/FABtotum/FABlin/releases/tag/1.0.0098.1

closed 2018-04-04 by krios-fabteam