FABtotum/fabui-colibri #176: Nozel heigth calibration issue

florentmaye, 2017-06-11:
It seems the nozel heigth calibration doesn't work properly: what ever I put in, even sometime several milimeter far from the bed, when I start a print the nozel start against the bed, no filament can go outside and I hav to manually drive it away from the bed, else the print don't come out correctly.

Besides in the calibration process, it is indicated (mm) whereas the nozel movements are cm.

florentmaye, 2017-06-14:
Even after the 1.0 version, the problem is still here and messup all my print attemps!!

christopherwit, 2017-06-19:
Maybe this is just to easy, but are you sure you have set the probe to be used during homing in:
settings>hardware>homing (bottom option enabled?)

If probe is disabled for homing the bed will go to a predefined height which can be set in, well somewhere :) don't know exactly how right away.

florentmaye, 2017-06-19:
I've never touched this option. It was desactivated by default, so I activate itd, but it didn't change my problem. (Since I have have this issue, there is always a z calibration before printing; when I didn't has it, there were no z calibration before it)
Does probe homing overwrite calibration?

And this doesn't explain the cm/mm confusion during calibration process.
Besides I forgot to mention another problem I encountered with this: sometimes when I want to do the calibration, when I ask the bed to go away from the nozzle, it goes far away (few cm whereas it should be mm) and if I ask it to come closer, it still go away but this time only of few mm ...

This issue is critical, I haven't succeed any print (over 20!!) because of this issue: either I don't succeed to have the nozzel far from the bed, and then filamant are finally stuck because 4th axis partially cut them by forcing on them, either I'm to far from the bed and it isn't stuck correctly to the bed ...

christopherwit, 2017-06-19:
I feel your pain, and more or less I did go through the same problems(some
still not able to consistently reproduce the issue).

By my knowledge the probe homing setting is introduced somewhere during the
beta testing of colibri fabui.
And for me the probe wasn't used during print start because of the default
setting. After I changed it to enable it did work. I had the same issue
with going up and down when pressing only up, but if I remind correctly it
had to do with one specific value. I think it was 0.5, but not sure anymore.

About the magnification of the z-axis displacement; are you talking about
the jog panel? Since the new firmware there is a possibility to multiply or
divide the set offset by 0.1x, 1x and 10x.
If it is none of the above I have to let you wait for a response of the
fabteam.

florentmaye, 2017-06-20:
No it is in the calibration panel.

florentmaye, 2017-06-20:
By the way, when activating probe during homing, the z calibration is done 2 times when printing: one before nozzel heating and one after ...

daniel-fabteam, 2017-06-21:
@florentmaye check if you don't have G28 in your start_gcode. There should be no homing code there.

florentmaye, 2017-06-21:
Yes there is one:
`;FLAVOR:RepRap
;TIME:102480
;Filament used: 9.90646m
;Layer height: 0.06
;Generated with Cura_SteamEngine 2.5.0
M104 S200
M109 S200
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
;LAYER_COUNT:278
;LAYER:0
M107`

I'm deleting it to see if it make the issue desappear.
It is strange, I'm using the configuration file provided for cura. I don't think I did such modification ....

florentmaye, 2017-06-21:
After investigation, I correctly have 
` "machine_start_gcode": {
      "default_value": "G21 ;Metric value\nG90 ;absolute positioning\nM82 ;set extruder to absolute mode\nM201 E100;\nM204 S1000 T80;\nM205 E5 ;\nG4 S1 ;1 millisecond pause to buffer the bep bep \nM728 ;FAB bep bep (start the print, go check the oozing and skirt lines adesion) \nG4 S1 ;1 second pause to reach the printer (run fast)\nG1 X10 Y10 Z70.0 F{speed_travel} ;move to oozing point \nG92 E0 ;zero the extruded length \nG1 F200 E50  ;slowly extrude 50mm of filament to clean the nozzle and build up extrusion pressure \nG92 E0 ;zero the extruded length again \nG1 F{speed_travel} ;Set travel speed \n;print"
    },`

in  _fabtotum.def.json_ but when lookin in Cura _Machine Settings_ for Fabtotum, I have in Start Gcode:

```
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
```

florentmaye, 2017-06-22:
I replaced both starting and ending geode.  There are no homing anymore at the beginning, it seems to follow the usual workflow. I did recalibration process then. 
However, I don't know if it is linked to this issue, but my filament never stick to the bed anymore. I have added hairspray, use heating bed, but everytime the filament don't stay on the bed. 

Any idea of what can cause this issue?

krios-fabteam, 2017-06-22:
We will correct _fabtotum.def.json_ to prevent any other issues with that
Can you confirm the issue about the nozzle height calibration is fixed by removing homing in start gcode?
About the filament adhesion are you using PLA or ABS? If you are using PLA usually a well-calibrated bed and hairspray are enough. Otherwise, if you are using ABS you can improve the adhesion to the bed by using some stickers.
You can take a look [here](http://www.fabtotum.com/improving-3d-printed-object-adhesion-tips-tricks/) for more tips and tricks

florentmaye, 2017-06-22:
I can't say as the print don't work.
It is black carbon PLA. Maybe the calibration isn't yet working correctly
and cause this issue...

<http://www.fabtotum.com/improving-3d-printed-object-adhesion-tips-tricks/>

florentmaye, 2017-07-01:
The calibration is still buggy (sometime don't move in the good direction ...). I don't know how it impacts the calibration process but I haven't been able to make a single print for more than a month, and it seems to be due to this issue:
    - nozzel too close => no filament get out of it and finally blocked because feeder "eat" the filament!!
    - nozzel too far => no adhesion to the bed ...

closed 2018-04-04 by krios-fabteam