FABtotum/fabui-colibri #177: Spool calibration

florentmaye, 2017-06-11:
First of all, when calibrating the 4th axis, the nozel isn't heated, which is a bad surprise the first time we calibrate.
Besides the calibration seems wrong: I tried to calibrate by measuring 10cm of filament, I got a value of 150mm, which doesn't work at all for me (nothing get out of the nozel when printing, whereas I can extrude filament in jog mode ...). The correct value seem more to be around 1500mm, even more in my case (I spent yesterday nigth fail-testing that ...)

florentmaye, 2017-07-01:
If I follow correctly the calibration process, almost no filament get out of the nozzel due to too low values! It's horrible I have to test manually values which work ...

rayyash, 2017-11-15:
I'm having a similar issue with printing, filament seems to be oozing out when I start to print but once the head actually gets on the bed nothing comes out. I'll check to see if it's a nozzle calibration issue.

closed 2018-04-04 by krios-fabteam

florentmaye, 2018-04-04:
Not closed : I retested  this weekend and it's absolutely not working. If I
follow the process I get a value 10 times smaller than the one which really
work.
Do you also have this issue too on your side ?

krios-fabteam, 2018-04-04:
Hi @florentmaye, 
just to be sure and aligned you are talking about "Step Calibration" under "Feeder" menu, right?
Could you paste here M503 output?

florentmaye, 2018-04-04:
Yes exactly.

Here is the M503 output :

*M503FABtotum TOTUMDUINO Batch Number: 5FABlin Version: V 1.1.1.1 Baudrate:
250000Steps per unit: M92 X72.58 Y72.58 Z2133.33 E1600.00Maximum feedrates
(mm/s): M203 X250.00 Y250.00 Z15.00 E12.00Maximum Acceleration (mm/s2):
M201 X10000 Y10000 Z50 E100Acceleration: S=acceleration, T=retract
acceleration M204 S4000.00 T100.00Advanced variables: S=Min feedrate
(mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms),
X=maximum XY jerk (mm/s), Z=maximum Z jerk (mm/s), E=maximum E jerk (mm/s)
M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00Home offset (mm): M206 X0.00
Y0.00 Z0.00PID settings: M301 P20.00 I3.50 D30.00Servo Endstop settings: R:
25 E: 127Z Probe Length: -32.00Installed head ID: M793 S2ok *

krios-fabteam, 2018-04-05:
Try this:
Remove the filament from the head
- go to "Maintenance > Feeder > Step Calibration"
- in "Quick settings" section click "Change value" button
- in the popup form insert "1524" and then click "Save" button
- now start with the calibration

After this do you have any luck?