FABtotum/fabui-colibri #179: Ethernet interface do not work

PaoloP74, 2017-06-13:
During installation of Fab-UI Colibri version 1.0, the ethernet interface can not be reached from Windows 10 home (version 1703).
The ethernet works with the Fab-UI version 0.987.

krios-fabteam, 2017-06-14:
Hi @PaoloP74, 
could you confirm you flashed a new sd card by following the [installation guide](https://github.com/FABtotum/fabui-colibri/wiki/Installation), for the official release a simple update from "Update" page is not enough

PaoloP74, 2017-06-14:
Yes. I deleted all SD partitions with "MiniTool Partition wizard" and created a single FAT32 partition. I copied the contents of the sdcard_latest.zip file and inserted the SD into the FabTotum.
The legs flash yellow and do not become blue.
With earlier versions I did not have this problem. I will try with a different SD.

krios-fabteam, 2017-06-14:
Did you also set "BOOT" as label for the FAT32 partition?

PaoloP74, 2017-06-14:
All ok. SD is broken. The other SD work.

closed 2017-06-14 by PaoloP74