FABtotum/fabui-colibri #181: probe angle range

christopherwit, 2017-06-14:
What has changed in probe extend and retract angles, and why? Previously it was working OK.

Now my issue, clearly the extend angle value is limited to 169(<170). You cant set any value above 169.
Previously my probe was perfectly vertical with a value of 127, but now it needs a value of more than 170 to make it deploy 90degrees.
With 169 the probe gets close to vertical but needs a few degrees still. I would make a well estimated guess a deploying value of 185 would allow me to get it completely vertical.

Please fix this issue once more and promise you wont try to improve things in future when it is already working.

infinity0n3, 2017-06-15:
Which part did you update? FabUI bundle or the firmware?

daniel-fabteam, 2017-06-15:
Which part did you update? FabUI bundle or the firmware?

christopherwit, 2017-06-15:
Everything, fabui, Fablin all to most recent version.

krios-fabteam, 2017-06-15:
This looks like more a related hardware issue. We didn't change anything about probe's angles,  if is needed to set angles bigger than 165 it means that the probe is not mounted as it should.  
Please contact support team by opening a [ticket](http://www.fabtotum.com/tickets) they will help you fix this issue

closed 2018-04-04 by krios-fabteam