FABtotum/fabui-colibri #182: Nozzle not heating after update

brettdaniels, 2017-06-14:
Just updated to colibri and I haven't been able to heat the nozzle since.

I've checked for damage, ran the first setup, but the only issues are the nozzle not heating and the nozzle height calibration stalling midway through, which is most likely from the nozzle not heating.

Some weird thing happens after I do a fresh wipe and copy of the latest archive. I boot up fine, with it indicating all the current firmware and software is up to date... but 10 minutes later without any update or flashing, the firmware page can't recognize what version it is, who made it, or when.

I don't really know what the issue could be... I don't see any damage at all to the cable.

krios-fabteam, 2017-06-14:
Did you update the firmware to the latest version? It is necessary for the new system to work

brettdaniels, 2017-06-14:
It shows firmware 1.1.0-1 when I go to firmware page, and in update info it says I have the same version as remote. Am I missing something?

krios-fabteam, 2017-06-15:
You have the latest version installed.
What head are you using? Did you installed it by going to "Maintenance > Head"? Are you able to move the head by using Jog panel? 
Could you also paste here "M503 Output" ? You can find it in "Maintenance > System Info"

brettdaniels, 2017-06-15:
I am using the Print Head V2, which I think is now known as Print Head Lite.

I did reinstall it by going to "Maintenance > Head"  and I am able to move the head using Jog panel.

Here is the M503 Output:

Steps per unit
M92 X72.58 Y72.58 Z2133.33 E3048.16
Maximum feedrates (mm/s)
M203 X250.00 Y250.00 Z15.00 E12.00
Maximum Acceleration (mm/s2)
M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration
M204 S4000.00 T100.00
Advanced variables
M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm)
M206 X0.00 Y0.00 Z0.00
PID
M301 P20.00 I3.50 D30.00
Servo Endstop
R: 40 E: 143
Z Probe Length
-38.00
Installed head ID
2
Batch number
1
Baudrate
250000
FABlin version
1.1.0-1

I keep hearing this periodic whirling sound that builds up to a what I think is a fan spinning. I think it is coming from the power supply. I don't know what it is but I wasn't hearing it at all before the issues showed up.

krios-fabteam, 2017-06-15:
Happy to hear that, maybe something was messed up on firmware.

closed 2017-06-15 by krios-fabteam