FABtotum/fabui-colibri #187: "Build plate needs to be flipped to the printing side"

brettdaniels, 2017-06-19:
Replaced my blue tape this morning and went to level the bed... got this error message:

"Build plate needs to be flipped to the printing side." 

It is on the printing side... wth is causing it to think otherwise? It won't let me continue to level the bed... help please

closed 2017-06-19 by brettdaniels

dmnc-net, 2017-07-06:
@brettdaniels How did you solve this problem. What this error even means?

dmnc-net, 2017-07-08:
@brettdaniels How did you solve this problem? What this error even means? Bed has to be placed with the glass on the top? Yes, it is, I can even level the bed in the Jog page but I can't calibrate it.

@simone-fabteam

krios-fabteam, 2017-07-10:
hi @dmnc-net, 
please make sure bed is correctly connected to the togo pins
could you paste here `M744` response? (Please use MDI in Jog panel)
If the bed is correctly connected you should get `TRIGGERED` otherwise `OPEN`
Could you also paste here `M503` output?
We look forward to your feedback.

dmnc-net, 2017-07-11:
Hi, thanks for your reply. I found the issue, calibration screws was too high so there was s problem with pins. Now calibration works. Thanks