FABtotum/fabui-colibri #190: Fresh install not working anymore

mgcrea, 2017-06-26:
I've updated my Fabtotum after not using it for a couple of months (was in 0.6), update went through but had annoying `Warning: cannot extrude filament: the nozzle temperature is too low` errors a few seconds after every print start (everything seemed ok though).

I decided to reflash a new SD card and the PI won't boot up anymore (I've followed the tuto and done this a few times already in the past with no issues whatsoever). Most troubling part is that I don't get any HDMI output on the PI when I turn it on.

Have you somehow changed the latest image PI support? (Like switching to a Pi3 arch?)

daniel-fabteam, 2017-06-26:
@mgcrea Hi, I just downloaded and flashed the new sdcard content. It booted without any problems.

If no HDMI signals is available it can mean that some boot files are missing. 
1. Either the sdcard image is corrupted. You can check this with `sdcard-v20170619.zip - md5sum: 52fbc6098a2f07806143ccc015297475`
2. The sdcard is faulty and corrupted. Can happen after many many re-flashes.
3. Some files skipped to be copied. (`kernel.img`, `start.elf`, `fixup.dat`, `bootcode.bin` those are mandatory for the boot process )

Try to format the sdcard again and copy the sdcard files to it. If that does not help, maybe try a different sdcard.

As we are out of beta phase no one is expected to flash the sdcards to get the latest updates working.
Next time you have an issue with the latest versions post an issue and we will try to help as soon as possible.

Hope this helps.

mgcrea, 2017-06-26:
Hi daniel, where can I find the latest "stable" release? The link in the README always point to the latest master:

Two consecutive downloads:

```
λ ~/Downloads
$ sha1sum sdcard_latest.zip 
1380ca44901a318a0e6e8ab4cc77522f5aa63a1a  sdcard_latest.zip
λ ~/Downloads
$ sha1sum sdcard_latest\ \(1\).zip 
1380ca44901a318a0e6e8ab4cc77522f5aa63a1a  sdcard_latest (1).zip
```

daniel-fabteam, 2017-06-26:
Use `md5sum` not `sha1sum`.

But yes, it's the correct one as I also get `1380ca44901a318a0e6e8ab4cc77522f5aa63a1a` when I use `sha1sum`.