FABtotum/fabui-colibri #191: Warning: cannot extrude filament: the nozzle temperature is too low

mgcrea, 2017-06-26:
Since I've updated from 0.6 to 1.0+, I can't print anything (original Indiegogo with Print head v2).

Print starts (correctly, nozzle temps stays around 200°C, does extrude) and stops ~15 secs later with the following error:

[image]

```
Warning: cannot extrude filament: the nozzle temperature is too low
```

The printer has been working quite well with previous colibri's version for a couple and was printing fine.

I did try a fresh install (even swapped the old RPi1 for a RPi3) but still got the same error.

```
Serial number
D3AC5-FAB-99E40
Os
Linux FabTotum 4.4.38-v7 #1 SMP Mon May 22 18:35:16 CET 2017 armv7l GNU/Linux
Fabui
1.0.3
Firmware
1.1.0	- Build date: May 29 2017 09:44:25	- Author: FABteam
Hardware
1
Installed head
Printing Head Lite
```

```
Steps per unit
M92 X72.58 Y72.58 Z2133.33 E3048.16
Maximum feedrates (mm/s)
M203 X250.00 Y250.00 Z15.00 E12.00
Maximum Acceleration (mm/s2)
M201 X10000 Y10000 Z50 E100
Acceleration: S=acceleration, T=retract acceleration
M204 S4000.00 T100.00
Advanced variables
M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm)
M206 X0.00 Y0.00 Z0.00
PID
M301 P20.00 I3.50 D30.00
Servo Endstop
R: 25 E: 127
Z Probe Length
-38.00
Installed head ID
2
Batch number
1
Baudrate
250000
FABlin version
1.1.0
```

Could you please at least add a checkbox in the safety settings to disable this buggy check?

krios-fabteam, 2017-06-27:
hi @mgcrea,
you should update the firmware (_Maintenance > Firmware_), your unit is running version `1.1.0` and the latest available version is `1.1.0-1` (with this the latest version we introduced some changes about temperatures reading ). 
If this not fix the issue please report it

mgcrea, 2017-06-27:
@krios-fabteam Just tried latest firmware and I'm still having the issue.

[image]

krios-fabteam, 2017-06-27:
Seems more like a hardware issue related to the head... 
Go to ` Maintenance > Nozzle > PID tune` and start the operation, this will calculate new PID values for the head. 
Let's see if this will solve your issue

mgcrea, 2017-06-27:
Hum still not better:

PID tune:
[image]

Following print:
[image]

Since it did work correctly before update and that the temperature appears to be totally OK, wouldn't this more qualify as a software issue?

krios-fabteam, 2017-06-27:
Did you saved the new  PID values by clicking on `Save` button? If yes in M503 output you should have:
```
PID
M301 P29.65 I4.27 D51.49
```
I would like to qualify this issue as software so it will be easier to fix but for what i see seems that the head can't hold the temperature set by gcode's file (`8 Bit Heart - 200°`) . Printing task is driven by what is on gcode file. 
Could you check if temperature is changed in gcode file (you can see it in `Project Manager`)?

[image]

Do you have the same issue also with other files?
Could you check also what is the minimum temperature in head settings ? (`Maintenance > Head` )
Standard value is `175°`, under this value the error is triggered to prevent any problems with the head and the feeder

[image]

mgcrea, 2017-06-29:
1. Yes I saved the PID values. It looks like the head does not like it.
1. Since I reflashed the SD-card, I had to re-download samples, they are in their original state:
```
M190 S40.000000
M109 S200.000000

;start gcode
G4 S1 ;1 millisecond pause to buffer the bep bep
M728 ;FAB bep bep (start the print, go check the oozing and skirt lines adesion)
G4 S1 ;1 second pause to reach the printer (run fast)
G1 X10 Y10 Z70.0 F15000 ;move to oozing point
G92 E0       ;zero the extruded length
G1 F250 E70  ;slowly extrude 30mm of filament to clean the nozzle and build up extrusion pressure
G92 E0       ;zero the extruded length again
G1 F15000 ;Set travel speed
;print

;Layer count: 258
;LAYER:0
M106 S255
```
1. Had the issue with one custom file (that used to work before), that lead to the whole reflash/reset thing. Also tried another sample with same results.

1. My head config seems OK.

[image]

--- 

I guess the actual temperature must not be properly available at all times thus triggering the error. The temperature clearly never reaches <175°C so maybe there is a wrong reporting (like null value) triggering the error?

- Did try lowering the min temp value to 100, and tried another sample. Somehow the print did work longer (thought it was fixed), but after a couple of minutes, it failed with the same message.

mgcrea, 2017-06-29:
Something strange, I did another PID tune today and it somehow doubled all variables values compared to the first run.

[image]

Looks like the PID tuning might not be working correctly. Will try to run another one right away.

mgcrea, 2017-06-29:
```
M502
M503
```

Did fix it. It looks like that I have to do this every time I reflash my SD-card. Otherwise the HEAD is not properly configured.

As stated before it looks like you should reset the head configuration (502) after every head change.

Here is the current 8-bit print after an M502. It looks like I get some heavy dips. Will try to re-do a PID-tune after this print (so far looks good).

[image]

mgcrea, 2017-06-29:
After multiple test my flat ribbon from the carriage to the printer might be damaged, not properly connected at all times, hence the issue maintaining proper temp.

Do you know if it can be swapped without replacing the whole carriage?

krios-fabteam, 2017-06-30:
You should contact our support by opening a [ticket](http://www.fabtotum.com/tickets/), they will help you

closed 2017-07-23 by mgcrea