FABtotum/fabui-colibri #192: RaspPi Camera Check-In

brettdaniels, 2017-07-02:
Is there any way to use the RaspPI Camera while the print is running? Last I checked it wasn't accessible during a print, but checking mid-print would be my only real need for this.

Could things be modified to allow this? I haven't found a use for the camera yet.

AndrewKillian, 2017-07-26:
I asked this a while ago and it was kinda shut down as the location of the camera makes it so it wouldnt be useful for the first couple centimeters of the print.

rayyash, 2017-11-17:
On the RP1a there's another slot you can attach a second camera, that was something I always had in mind. At the moment I'm using an old android phone as an ip Camera to check in for long projects.

closed 2018-04-04 by krios-fabteam