FABtotum/fabui-colibri #195: almost no filament get out of the nozzle

florentmaye, 2017-07-23:
It's now more than 2 month I can't print anything. Currently when I try to start a print the head move correctly but no filament get out of it. I tried to put the head at 1cm of the plate to check if filament can get out, and indeed, almost nothing get out of the nozzle. 
My printer is calibrated (in fact even over calibrated as I set 30cm get out of the nozzle when calibrate for 10). I can correctly extrude in jog mode, calibration mode and with print example loaded, but no with customs one. 
This make me think that it could come from slicing profile. I'm currently using the lastest version of the slicer profile you released, with Cura 1.6. Here is the header of my gcode. Does it seems correct for you ? I'm not sure but it seems incorrect for me ...

`;FLAVOR:BFB

;TIME:88420

;Generated with Cura_SteamEngine 2.6.0

M104 S200

M109 S200

G21
G90
M82
M201 E100
M204 S1000 T80
M205 E5
G4 S1
M728
G4 S1
G1 X10 Y10 Z70.0 F120
G92 E0
G1 F200 E50
G92 E0
G1 F120

;enable auto-retraction

M227 S16640 P16640

;LAYER_COUNT:305

;LAYER:0

M107

M103

G1 X156.343 Y47.388 Z0.3 F3600

G1 X156.343 Y47.388 Z0.3 F3600

;TYPE:SKIRT

M108 S22.5

M101

G1 X168.51 Y48.59 Z0.3 F1800`

AndrewKillian, 2017-07-26:
I've been having this issue too I've cleaned the nozzle extensively so I'm fairly certain it's not a mechanical problem unless there's something wrong with the feeder. My issue is that it will print the perimeter shells fine but the infill barely comes out and is all stringy.

florentmaye, 2017-07-27:
> So I've tried slicing with Simplify 3d and that fixed my infill issue.
> Does FABUI change the gcode at all when you upload it? Maybe it's changing
> things in cura or maybe it's the profile I don't know but I'm fairly
> certain it's not the machine itself.

If Fabtotum team is willing to offer me Simplify 3D, I'm ok with this ^^,
otherwise that could be cool to correct this asap!!

closed 2018-04-04 by krios-fabteam