FABtotum/fabui-colibri #199: Current sd_card latest does not boot anymore

mgcrea, 2017-09-01:
It get stuck at `random: nonblocking pool is initialized`.

Always worked flawlessly in the past. It's a RPI3 (retrofit on a Indiegogo FabTotum).

```
$ sha1sum sdcard_latest.zip 
10cf6f2acff3892cba596176933109489db06968  sdcard_latest.zip
```

daniel-fabteam, 2017-09-01:
Hi Olivier

Do you get stuck before the `firstboot`, `after` it or after a `reboot`?
Which address do you set in `earlyboot.conf` ?

m-sja, 2017-09-03:
Hi, i have the same problem.
After a corrupted SD-Card i got a new 8g SD-Card, formated it, copied the Files, put it in the Pi (1A), powered up.
It is stuck on:

`== First Time Boot ==`
`uart-p1011 ......`
`smsc95xx ... link up....`
`TOTAL_TASK_COUNT 36`
`random: nonblocking pool is initialized`

LEDs keep fading green.

IP-Config is not changed `169.254.1.2`.
I configured my ip to `169.254.1.3`
Link is up but no response on when i open the printers IP in Chrome.

I can't find old stable Builds for download...

krios-fabteam, 2017-09-04:
Hi guys,
we have just released a new update, could you download the new [sdcard](http://update.fabtotum.com/colibri/armhf/images/sdcard_latest.zip) and try with it?

m-sja, 2017-09-04:
Works for me, thanks!

Btw., an archive of published images would be nice.
Version numbers in the filename is also a good thing.

Thanks and best regards
Marian

krios-fabteam, 2017-09-05:
Actually, there's one list available [here](http://update.fabtotum.com/admin/list.php)

closed 2017-09-06 by krios-fabteam