FABtotum/fabui-colibri #200: Not working

Fensterbank, 2017-09-09:
Hello,

I followed the wiki to format SD Card and sucessfully went through the first installation web interface until the system wanted to restard (with this 60 seconds timer).
But: Never any light went red, blue or whatever. It blinks yellow like ever. I also doesn't beep. It makes nothing with the Fabtotum.

Afther the timer passed, nothing happened.
I connected my monitor and realized, lighttpd wasn't even started. I rebooted the system.
When I watch Colibri starting on monitor all looks fine, lighttpd ist starting, FabtotumServices.py is running, ethernet is working...
but, the system is dead. When I open the IP address in browser, it is loading (so lighttpd listens to the port) but nothing happens.

When I reboot, I see
`[FAIL] Stopping FABUI WebSocket Server`

During boot Starting the WebSocket Server succeeded, so I think this thing crashed.

I have the first Fabtotum from the crowdfunding. Is this thing even compatible with my Fabtotum model?

daniel-fabteam, 2017-09-11:
Hi Frédéric,

The new FABUI is compatible with all units out there.

However there are a lot of decompression errors in the logs you attached.
`SQUASHFS error: lzo decompression failed, data probably corrupt`

Files on the sdcard are corrupted so try to flash the sdcard again or even try a different sdcard.

Fensterbank, 2017-09-11:
Thanks for the response.
I'll try it again asap and give feedback here.

dr4Ke, 2017-09-20:
For what it's worth, I had the same issues during the beta phase. I had to give up the original fabtotum sdcard because there was too often corruption on it.

Fensterbank, 2018-03-20:
So I want to give it another try.
Loaded the latest sd_card.zip (`sha1sum ab6ebd6d437ea4e2ff7c2a39e4c30a0b21342289`), extracted all files (tried two different tools and the `unzip` in linux command line) to a new 16 GB sd card.

But now it just got stuck. I delayed the boot time in the config to make sure it's not hanging at `random: nonblocking pool is initialized` (https://github.com/FABtotum/fabui-colibri/issues/199) and indeed it stops on `First Time Boot` after having the network. I also tried it without network. Doesn't change anything.

The IP configuration is correct and I can even ping this machine. But that's all.
[image]

Also it's not freezed. I can type in things and when disconnecting or connecting network or usb devices there are messages on the screen... But whatever is in your script, it does not run.

I don't know what I can do anymore... I mean, it's the original Fabtotum from Crowdfunding with the original Raspberry Pi. It should work!
I'm about to throw the whole thing in the trash and declare the Fabtotum project dead for me. :(

Seriously, does anyone sell those working sd cards? Can I send my Fabtotum somewhere to fix that? I'm at my wits' end.

Fensterbank, 2018-03-20:
Indeed, it worked with another SD card.
There was a bit confusion (maybe someone received a mail from github with another content) because nothing happened on screen and while I thought it's stuck again, there was the web interface just waiting for me to start the installation.

Maybe a boot message like "please open 192.168.1.50 to continue the installation" would be helpful. 😉

closed 2018-03-20 by Fensterbank