FABtotum/fabui-colibri #201: Install on clean SD-Card failed after Reboot

m-sja, 2017-09-11:
After formatting the SD-Card and unzipping the files to it, i run the initial Setup.
(Language, Account, etc ...)

During Install i had HDMI connected to the RPi1a.
I got the following error messages:
[image]
[image]

In the UI setup process there is an error (i tried Englisch and German).

On step 4 the dialog refers to the "Install"-Button, but there is none.
The only button wich works is "Next", but the cursor is changing to a "Forbidden" Symbol.
But it works ?confused?
[image]

The Setup process finishes and the Printer is usable until reboot.
Then i got these errors leading to kernel panic:
[image]

I tried two different Pi1a with two SD-Cards, Micro-SD-Card and one Pi1b. - No Luck
I tried all the available fabui-colibri zip-files from [here](http://update.fabtotum.com/admin/list.php) - No Luck
It seems that fabui-colibri is running so well and stable, that nobody gets to the point where i am now ;-)

Thanks for Help.

m-sja, 2017-09-11:
Btw.
This is the storage-information page after the initial Setup.
It seems there are major issues regarding partitions:
[image]

infinity0n3, 2017-09-12:
The partitions were not created at all. Did you follow the instructions for formatting the sdcard?
[SDCard Formatting](https://github.com/FABtotum/fabui-colibri/wiki/SD-card-formatting)

What OS do you use to format the sdcard?
What is the size of your sdcards?

daniel-fabteam, 2017-09-12:
The partitions were not created at all. Did you follow the instructions for formatting the sdcard?
[SDCard Formatting](https://github.com/FABtotum/fabui-colibri/wiki/SD-card-formatting)

What OS do you use to format the sdcard?
What is the size and type (SD, SDHC, SDXC) of your sdcards?

m-sja, 2017-09-12:
Hey Daniel, thanks for the hint.
I'm using ubuntu mate 16.4 lts.
I formatted the SD-Cards as i always did (deleting all partitions, creating a new W95 FAT32 c, boot flag).
But i was not aware, that there is a "Special" way to do it. It always worked except for the issues above.

I tried the gui (Linux) method step by step and got an error while formatting the new created Partition.
`Error synchronizing after initial wipe: Timed out waiting for object (udisks-error-quark, 0)`
The solution was to change the card-reader! The internal reader of my thinkpad seems to cause some issues. A generic cheap reader solved the issue.

One 16gb Samsung SDHC is working now. :+1: 
The original "FABtotum" 8gb (wich came with the Printer) formats and installs fine, but is still giving errors mounting "bigtemp" from /dev/mmcblk0p5. The other partitions are fine. I assume the Card is dead somehow.
With my 8gb SanDisk Ultra micro-sdhc the RPi refuse to start if i format your way.

It's kinda weird, but at least i have one working card.

closed 2017-09-12 by m-sja