FABtotum/fabui-colibri #214: Vagrant box no longer available

LuxAetern, 2018-12-10:
The vagrant box in the vagrant file is no longer available as it was hosted on fabtotum.com
Can someone publish said box on vagrant again?
Or at least tell me what was installed so that i can recreate it under opentotum?

I am also missing the documentation on how to compile everything..
Can somebody create a Readme or similar?

PaoloP74, 2018-12-10:
I do not know if it's up to date.
--> https://app.vagrantup.com/fabtotum/boxes/centosdev

visualinventor, 2019-05-28:
Did you ever get a vagrant box setup? I too, am looking to add to/change the repo?

PaoloP74, 2019-05-29:
--> https://app.vagrantup.com/fabtotum/boxes/centosdev

CryoRig, 2019-05-29:
@visualinventor unfortunatly not
@PaoloP74 i already know about that link, but the problem is that the original Image which vagrant used is no onger available, as that was hosted on fabtotums own webserver

PaoloP74, 2019-06-20:
Posted by Marco Rizzuto on FB: https://www.dropbox.com/s/gfg7mrzq1v1z49f/centosdev-0.0.1.box?dl=0&fbclid=IwAR0pUeM2ef11cr9gp8UB3q_MOLeDJ0UxW01AsrEl9O4LPJsHm9NZbrZiJeQ

closed 2023-05-08 by CryoRig