FABtotum/fabui-colibri #216: Problems installing

cmuscata, 2019-04-15:
Hi everyone, haven´t used my Fabtotum in a couple of years.  Tried following all the steps mentioned in the wiki to install the software.  Bought a new SD card as everyone says they had problems with the original SD CARD (well it´s a microSD in an SD adapter.)  Formatted with MiniPartition tool on Windows 7.

Installation seems to go fine but the lights just remain green and the dashboard won´t connect.  Is there a log file I can screenshot and post here for more info?  Not sure what might be wrong.

[image]
Thanks, any help really appreciated.

infinity0n3, 2019-04-15:
Hello, I has been a while. There is a plugin that collects the logs after which you can download them as a zip file if I remember correctly. The link is [here](https://github.com/FABtotum/fab_system_logs/archive/master.zip). Download the file and go to plugins sections. There should be an upload option somewhere. You can install the plugin like that.
After that you should be able to activate it and export the logs.

cmuscata, 2019-04-15:
Hmmm sorry about this but i´m not too technical...unzipped the download and i see these folders n files: 
[image]

don´t see any plugins folder or anything I can install.  Is there meant to be a .exe or smthg?

infinity0n3, 2019-04-15:
Ok, no problem. Do not unzip the file.

First try to see if the "System logs" plugin is not already there. If I remember correctly it should be installed by default.
Go to plugins and see if it is there. Maybe it needs to be activated first.

If not, then follow the instructions below.

In FABUI there should be "Plugins" among menu items on the left. 
1. When you click on it there will be a tab that is named "upload". 
2. There you can click on the upload button and select the downloaded zip file.
3. After that (not sure anymore) if everything goes right in the plugins list tab there should be a new one called "System logs". 
4. Maybe you will need to activate it first, sorry I don't remember it anymore (even though I made it xD)
5. Now there should be a new menu item "System logs" where things should be self explanatory.

cmuscata, 2019-04-15:
Nah don´t worry, really appreciate your help.  Doesn´t seem to work :( It just gets stuck on "uploading and installing..."

infinity0n3, 2019-04-15:
Not sure what is going on especially without logs.

But at the moment my best guess is that many the firmware is not compatible with the fabui version you have installed as non of the firmware related readings is visible. (n.a.)

Try the "Flash firmware" button with "Factory default" selected.

Otherwise you could try to flash some of the [Marlin.cpp.hex](https://github.com/FABtotum/FABlin/releases) files. Insted of "factory default" there should be an option with "Custom..."

cmuscata, 2019-04-15:
IT WORKED!!! Thanks buddy, you´re a star.

infinity0n3, 2019-04-16:
I'm glad I could help