FABtotum/fabui-colibri #218: Fabui upgrade raspberry Pi4?

MIRCO3TEM, 2020-01-02:
Buon giorno

Vorrei sapere se esiste una versione di Fabui compatibile con il Raspberry Pi 4 o se sono obbligato a fermarmi al Raspberry Pi 3 

Grazie per le eventuali risposte

infinity0n3, 2020-01-02:
Hello Mirco,

There is no of fabui v1.0 version compatible with RPI4. 

However a new [FABUI](https://gitlab.com/colibri-embedded/fabui/fabui-frontend/tree/develop) version compatible with FABtotum and other printers is being developed.

If you want I will include you in beta testing once it starts. The new version will be compatible with RPI4 in the future.

PaoloP74, 2020-01-04:
Hi @infinity0n3, I would like to participate in the beta to test the new FabUI with the FabTotum and the Marlin firmware.
I think it is possible to switch from FabLin to Marlin 2.0.x. but I have not carried out tests as it is blocked by the ugliness of the OctoPrint interface, the only alternative to FabUI.
I would therefore like to try the new interface that I assume uses the G code standard in order to interface it with the Marlin standard.

infinity0n3, 2020-01-04:
Hello Paolo,

Some time ago I did make patches for Marlin 1.1.5 that allowed it run on fabtotum and be able to print. 
I would like to port it to 2.0 or at least the latest Marlin 1.1.x but I just don't have the time.
In case you want or anyone else wants to give it a try, [here](https://github.com/infinity0n3/FABlin-ng) are the patches I made.

For the moment I can only offer access to the [alpha version](https://gitlab.com/colibri-embedded/fabui) of FABUI 2.0.

It's the same procedure as with FABUI 1.0, copy the sdcard image files to a FAT32 formatted sdcard (best use 8GB) and give it a go.
There is a step where you choose the printer. Select FABtotum and Core Pro. At the moment that is the only one in the list. The Generic Marlin options are not functional yet.

If you would give me a description of which FABtotum unit you have I can add more versions to the list with the correct initializations. For the moment there is no automatic FABtotum unit version detection.

PaoloP74, 2020-01-05:
Hello,
the version of my FabTotum is 2500, it was a Fabtotum Core updated to the Fabtotum Core Pro.
During the update they removed the camera and the fourth axis.
This is the outcome of the M503 command:

```
Steps per unit M92 X72.58 Y72.58 Z2133.33 E91.00
Maximum feedrates (mm/s) M203 X250.00 Y250.00 Z15.00 E300.00
Maximum Acceleration (mm/s2) M201 X10000 Y10000 Z50 E300
Acceleration: S=acceleration, T=retract acceleration M204 S2000.00 T525.00
Advanced variables M205 S0.00 T0.00 B20000 X25.00 Z0.40 E1.00
Home offset (mm) M206 X0.00 Y0.00 Z0.00
PID M301 P53.54 I9.08 D78.94
Servo Endstop R: 45 E: 146
Z Probe Length -32.00
Installed head ID 6
Batch number 2500
Baudrate 250000
FABlin version 1.1.1.4
```
I will gladly test the alpha version of the new fabui.

PaoloP74, 2020-01-05:
I created the SD as indicated and inserted it in the FabTotum instead of the original that I will keep as a backup.
Linux Colibrì boots, configures account, printer (is recognized) and network.
Then I press FINISH and the bar below remains in constant movement without doing anything.
I waited a few minutes but nothing happened. It doesn't go on.
If I connect via WiFi, the setup starts again.

infinity0n3, 2020-01-06:
Let's continue with new issues [here](https://gitlab.com/colibri-embedded/fabui/fabui-frontend/issues).

There is an updated sdcard image with should fix the setup issue.

MIRCO3TEM, 2020-01-13:
grazie per la risposta
quando ho tempo provo la volentieri la beta

quale versione uso su Raspberry pi 3+??
mi potete dare un link per effettuare il download?

PaoloP74, 2020-01-13:
@MIRCO3TEM La nuova FabUI è ancora in versione alpha. Se ti interessa aiutare nello sviluppo e segnalare bug puoi scaricare l'mmagine nel link che trovi nel messaggio quotato. 
La nuova versione è utilizzabile dal raspberry pi 2 o superiore.

> Hello Paolo,
>
> Some time ago I did make patches for Marlin 1.1.5 that allowed it run on fabtotum and be able to print.
> I would like to port it to 2.0 or at least the latest Marlin 1.1.x but I just [...]

infinity0n3, 2020-01-13:
Although I haven't tested 2B version. It will probably work. However there will be no wireless connection and USB dongles are not supported.

Version 3B or 3B+ will work.

Shaltz, 2020-05-10:
hi @infinity0n3 , 
I'd love to help you out with fabUI. 
I have an original (indigogo) fabtotum and I'm a linux sysadminwith a full stack dev background, if you fancy some help, give me a shout

PaoloP74, 2020-05-10:
Hi @Shaltz , you can find the new project [here](https://gitlab.com/colibri-embedded/fabui/fabui-frontend/-/wikis/welcome).

infinity0n3, 2020-05-10:
Hi @Shaltz,
It would help to get a feed back on current state of FABUI.
What is working for you, what is not so I can fix things.

Issue that you find report on the gitlab page.

Also, I'm working on adding milling and laser support to the new fabui.

Fisheiyy, 2022-04-23:
> Hi @Shaltz, It would help to get a feed back on current state of FABUI. What is working for you, what is not so I can fix things.
> 
> Issue that you find report on the gitlab page.
> 
> Also, I'm working on adding milling and laser support to the new fabui.

Is Colibri-Embedded here on Github up to date with Colibri-Embedded on GitLab? I am interested in helping out and im sure I can find a few others since it would bring back functionality and fix problems that people who still actually use Fabtotum printers have and deal with, im sure if you ask in the [Facebook Group](https://www.facebook.com/groups/1052715011528991/) if people are interested you will get quite a reaction. The Facebook Group is the only place I can think of where people still actually talk about and help others with/about Fabtotum printers.