FABtotum/fabui-colibri #220: Problem installing firmware

Morellid97, 2020-11-22:
Hi everyone, i recently bought a Fabtotum core, and after 4 months int started having problems, (I was unable to locate it's IP address) so I decided to flash the firmware on the SD card. After completing the steps on the fabtotum guide lines site I am stuck on compiling the marling.io file. It seems like it cannot find some .h files so it wont compile. After reputting the old files on the SD card it wont even reeboot now. Anyone can help me on that?

PaoloP74, 2020-11-23:
Hi, it is not necessary to compile any files.
You simply have to copy the files to a FAT32 formatted SD and insert it into the Raspberry SD slot.
You can find the latest release of the FabUI here: https://github.com/FABtotum/fabui-colibri/releases/tag/1.1.6
While the instructions for the SD can be found here: https://github.com/FABtotum/fabui-colibri/wiki/SD-card-formatting
The FabTotum IP address is 169.254.1.2

rayyash, 2020-11-24:
Just download and extract the fab ui files from the zipped folder then copy
them directly on the SD. No other steps are needed.

Morellid97, 2020-11-24:
Thanks guys, it works now!

closed 2020-11-24 by Morellid97